LVMakerHub/LINX #128: SPI communication with STM32 using linx

L1ikhith, 2024-04-17:
Hello,
i was getting errors with one of the SPI functions in LinxWiringDevice. The issue is  in LinxWiringDevice. cpp, the bitOrder in  SPI.setBitOrder(bitOrder)  is defined as unsigned char but in STM32 SPI file it was defined as Enum. Is there a way to make the function variable as Enum instead of unsigned char 

_REF:_
**SPI.cpp:**
`void SPIClass::setBitOrder(BitOrder bitOrder)
{
  _spiSettings.bitOrder = bitOrder;

  spi_init(&_spi, _spiSettings.clockFreq,
           _spiSettings.dataMode,
           _spiSettings.bitOrder);
}`
**Wiring_constants.h**
`enum BitOrder {
  LSBFIRST = 0,
  MSBFIRST = 1
};
`
**LinuxWiringDevice.cpp**
`int LinxWiringDevice::SpiSetBitOrder(unsigned char channel, unsigned char bitOrder)
{
	SPI.setBitOrder(bitOrder);
	return 0;
}
`

samkristoff, 2024-04-17:
Can you link to the SPI.cpp file you're referring to?

samkristoff, 2024-04-17:
Is this the BitOrder you're rerring to:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/cores/maple/wirish_constants.h

Can you ellaborate on the erorrs?  Are you building the C++ code and getting compile time errors, runtime errors, etc?

Thanks!

L1ikhith, 2024-04-18:
Hello @samkristoff thanks for the quick reply, here are the links to the files 
**SPI.cpp**: [https://github.com/stm32duino/Arduino_Core_STM32/blob/main/libraries/SPI/src/SPI.cpp](url)
**Wiring_constants.h** :[https://github.com/stm32duino/Arduino_Core_STM32/blob/main/cores/arduino/wiring_constants.h](url)

**ERROR screenshot**
[image]

samkristoff, 2024-04-20:
Hey, can you just cast the bitOrder to unsigned char on line 310 like this:

`SPI.setBitOrder((unsigned char)bitOrder);`

L1ikhith, 2024-04-23:
@samkristoff i think i cant able to cast the bitOrder to unsigned char, in the wiring_constant bitOrder was declared as enum and the SPI function was wrote in terms as enum as input

L1ikhith, 2024-04-23:
@samkristoff  i think i cant able to cast unsigned char for bitOrder, the function SPI.setBitOrder was designed based on enum as input for the function.

closed 2024-04-25 by L1ikhith