LVMakerHub/LINX #129: Trying to use Linx with STM32F4

L1ikhith, 2024-04-25:
Hello 
I am trying to develop custom firmware with linx to talk with STM32F446ZE board(BTT octopus V1.1). the board is working using custom firmware built using platform now i am trying to integrate linx to the firmware so i can control the board using labview, i was able to integrate linx with my firmware without any errors but when i tried to run a Manual Blink example from labview i was getting Error 5003.

here is the link to the project 
`https://github.com/L1ikhith/Linx-STM32f`

L1ikhith, 2024-05-01:
Hello @samkristoff @spathiwa  i had some success with the communication but I am stuck at Load Device config vi, 
Error: in the Load Device config all the modules are working until the LSB reaches 10, once it reaches 10 instead of sending all the response packet at once the data is splitting into multiple packets(see example) 

for example 
when the LSB is at 10 if the data size is 0 the response packet is  255, 6, 0,10,0,0 instead of 255,6,0,10,0,15 and in the second iteration its sending 0 and 15
i am not sure why its causing this issue  can you please let me know what's causing this issue

samkristoff, 2024-05-03:
A couple thoughts.  The UART buffer size on the device is defined [here ](https://github.com/L1ikhith/Linx-STM32f/blob/8a8bba4ff1120b6c831d1d1098d6a3984ac51262/STM32-LINX-LABVIEW/src/LinxSTM32F4.cpp#L70C1-L71C1).  This determines the max packet size since it has to fit in the buffer.  Yours looks fine.

Decimal 10 is LineFeed (LF) in ASCII.  I think the VISA driver uses LF as a delimiter by default but the LINX VI should disable that under the hood.

L1ikhith, 2024-05-03:
Thanks for the reply @samkristoff  i didn't modified any Linx vi's. "Decimal 10 is LineFeed (LF) in ASCII." this makes sense whenever the LSB or MSB bit reaches 10 its chopping the data into two 
for ex if the data packet  is[ 255,6,0,10,0,15] at 10 instead of sending the whole packet i am seeing [255,6,10,0,0] and [0,15,0,0,0] 
this issue is only happening on STM 32 board when i tried with Arduino Uno the data was not corrupted at bit 10

One more issue that i am facing was with set device baud rate Vi, the issue was during the execution of this vi it will first disconnect  the existing serial communication and reinitializes with the new baud rate right, when its reinitialize the communication instead of using the same COM port  a new alias resources was created with Port binding and initializing the communication with the new alias; this is only happening on STM 32 board there was no issue with UNO 

My guess was the port was not getting closed properly is there any way to make sure the port closes properly?

Note: the device was connected to COM 9
new alias were created every time when the serial was initialized 
[image]