LVMakerHub/LINX #131: Installed LabVIEW on an armbian embedded board, but cannot call the serial port

ericddm, 2024-07-19:
Dear developers and maintainers, 
I installed LabVIEW on **WuKong Pi**, a single board computer running Armbian, as shown in the picture:
But I encountered a problem. **It can only run vi that does not call the serial port.** Once the serial port is called, the LabVIEW service stops.
I wonder if I need to define this serial port somewhere?
Please help me, thanks

[image]
[image]

samkristoff, 2024-07-19:
Are you using the VISA VIs in LabVIEW to interact with the serial port?  If so, VISA needs to be installed on the target (WuKong Pi).  

Does LabVIEW throw an error when this happens?

ericddm, 2024-07-20:
Dear samkristoff, hello, thank you very much for your attention to this question.
First of all, I would like to add that this single-board computer can automatically run LabVIEW when powered on, and there is no problem running the blink program that does not require calling the interface.
But when I start calling the linx vi, even if I just open the device, an error pops up:
========
Load Device Channels.vi loaded with errors on the target and was closed.
========
[image]
[image]
[image]

samkristoff, 2024-07-21:
When using local IO, the VIs calls into a shared object under the hood and the shared object will make target specific calls.  It looks like your LabVIEW project is treating your target as a BeagleBone black so it may be trying to do something that works on BBB but not on your WuKong Pi.

@sharpk do you recall how the correct LINX shared object gets on the target?

ericddm, 2024-07-21:
Dear Sam, @samkristoff 
Thank you for your help and I look forward to your updates.
To make my problem clearer, I tried the following.
[1] Last time you said **VISA** needs to be installed on the target. I looked for the official NI visa driver, but there was **no one suitable for Debian**, as shown in Figure 1.
[2] Figure 2 shows my other attempts. **When using local IO, it is indeed stuck at initialization**.
[3] The Wukong Pi is a low-cost little single-board computer that costs about $10, and if you have the time to play with it, I would be relly happy to **mail one to you**.
[image]
[image]
[image]

ericddm, 2024-07-21:
I just tried adding RPI (actually Wukong Pi) to the LabVIEW project. It can connect and run Blink, but it is also unable to call local IO.
[image]
[image]

sharpk, 2024-07-23:
So everything, including the linx library (liblinxdevice.so) is installed by the debian package [here](http://feeds.labviewmakerhub.com/debian/binary/).

There are separate versions of liblinxdevice.so for Raspberry Pi and BBB, and at install time a script is run to detect what hardware the package is being installed on and a symlink is created to the correct library.

You will likely need to create a new version of the LINX library in order to do the correct thing for your board to get IO working correctly.  I doubt that the pre-compiled Raspberry Pi or BBB versions of the library will work on your hardware (although you can give it a try if you like).

ericddm, 2024-12-20:
Dear @sharpk @samkristoff , Very good evening. 
I'm trying to install LabVIEW on **WuKong Pi**. I didn't find the **liblinxdevice.so** file you mentioned in the github repository, and I was wondering if you could make the **deb file** public. 
I don't know if this request is too much, also not sure whether it will infringe your intellectual property rights, I just hold the idea of give it a try. I'll accept whatever you decide. 
**Thank you very much for your excellent work, linx is amazing.**

ericddm, 2025-02-02:
I haven't received your reply for a long time. In order to avoid forgetting, I repeat my requirements again. Liblinxdevice_rpi2.so and liblinxdevice_bbb.so are the source codes of these two files. Can two outstanding @sharpk @samkristoff provide them to me? 
Thank you very, very much.

samkristoff, 2025-02-02:
The source code for the shared objects is in https://github.com/LVMakerHub/LINX/tree/main/LabVIEW/vi.lib/MakerHub/LINX/Firmware/Source

See the makefile here: https://github.com/LVMakerHub/LINX/blob/main/LabVIEW/vi.lib/MakerHub/LINX/Firmware/Source/make/Makefile

ericddm, 2025-02-03:
> The source code for the shared objects is in https://github.com/LVMakerHub/LINX/tree/main/LabVIEW/vi.lib/MakerHub/LINX/Firmware/Source
> 
> See the makefile here: https://github.com/LVMakerHub/LINX/blob/main/LabVIEW/vi.lib/MakerHub/LINX/Firmware/Source/make/Makefile

thank you very much, @samkristoff